jneilliii/OctoPrint-MQTTSubscribe #1: Can't get this plugin to work

SoundGuy, 2018-10-10:
1. I couldn't find the plugin in the plugin manager, i installed it through the URL in the Readme.md 
2. I've set it up exactly like in the instuctions. with "connection" as  topic. 

PUblishing an MQTT to "octoprint/plugins/mqttsubscribe/connection" didn't result in any response from Octiprint. 
I didn't see any MQTT message sent back
and i didn't see any log in the octoprint.log

in both the MQTT message body and the settings 
I tried to enter either 
"{"command":"fake_ack"}" 
or  just 
"fake_ack"
with no positive results.

what steps can i take to see what i should be doing to make it work?

SoundGuy, 2018-10-10:
updarte. After changing settings i've tried to reboot octoprint and not it start to do something.

1. seems like the documentation isn't complete.
1a. there is a mismatch between "Message" and "Command"
2. the full topic is as follows:

octoprint/plugins/mqttsubscribe/connection/1

it MUST insclude the index

when i use post i get 204 and no info and no content
when i use get I get the whole json reply in the response.

jneilliii, 2018-10-12:
Your issue is mainly the reason this is not listed in the plugin repository/plugin manager.  It's very finicky and if you don't understand octoprint's api calls it's difficult to configure. If you give me an example of what you're trying to do specifically I can test/verify and give you a screenshot of the settings to use to get it to work.

SoundGuy, 2018-10-15:
I've sotra managed to get this plug in to work after a few attempts.

I got Alexa / Siri to send an MQTT command when i say "preheat PLA" with the temprature 200 
I've tried a few things:
[image]

I've managed to get the connection command to respond to the MQTT string with Fake_ack, but the connect command didn't actually connect to the printed.
I've managed to see that the various versions of the tool command were received by a pop up on the main screen of OctoPrint, but didn't actually warm the tool. 
I tried both 
{"command": "target",   "targets": { "tool": 200 } }
and even copied from the example:
{   "command": "target",   "targets": {     "tool0": 220,     "tool1": 205   } }
they are recived but not handled. the tool stays on 0.

eventually i got it to work by sending an GCODE comamnd instead 
{   "command": "M104 S200 T0;" }

now it turns on when i say "Hey Siri, preheat PLA" . I will make a video of that and post it, cause it is cool that i can do that handsfree, especially when i already have filament in my hand and forgot to press the dang button before i started putting it in.

i there could also be an option to parametrize it so the plugin won't match the whole string when it accepts the requests. 
but for now it's enough to me. I might help you extend the plug in if i need to in the future, seems like i'm the only person right now who is doing that :)

should I close this issue?

jneilliii, 2018-10-15:
That's awesome, glad it worked out. I'd be more than happy to accept pull requests. I think I'm going to focus my efforts on incorporating a inbound command topic to the mqtt plugin directly now that I have contribution rights to it.

closed 2018-10-15 by jneilliii