jneilliii/OctoPrint-MQTTSubscribe #3: Status code : 400

yaabaa, 2019-08-11:
Hi, 

I would like to command via mqtt my 3D print with octoprint, for instance this : {"command":"G28"}

I wrote this in payload, with json format in Node Red. 
Mqtt plugin and mqtt subscribe plugin installed and configured.
[image]

When I send command in node red, a pop up appear in octoprint screen with status code : 400.

Check status code definition [here ](https://www.w3.org/Protocols/rfc2616/rfc2616-sec10.html#sec10.2.1) but can't find solution.

Any idea to help me ?

Thank you,

jneilliii, 2019-08-11:
Can you post a screenshot of the plugin's settings?

jneilliii, 2019-08-11:
The Topic should be `printer/command` and the payload is `{"command":"G28"}`.

yaabaa, 2019-08-12:
Hi, 
Thank you for your quick reply.
I adjust topic as you mentioned, but still doesn't work.

Here some screen about plugin and node red mitt publish ( inject is set to json )

[image]
[image]

Thank you a lot for your help !

yaabaa, 2019-08-12:
Dear Jneilliii, 

I update / reboot my Pi, and it's working now. Perfect for my project.

Thank you a lot for your help and your time to develop this plugin.

Bye,

closed 2019-08-12 by yaabaa