jneilliii/OctoPrint-MQTTSubscribe #4: why the index thing ? does not match the documentation

aschor, 2019-11-17:
Hi, basically I fought as anyone to get this to work.

For getting something working, we have to NOT follow the documentation. it says

> Topic: topic to subscribe to. Should be the same as base OctoPrint REST API path, ie if API is /api/connection you use connection.

and the screenshot with it is similar .... they do NOT include any "index" value.

But to get the plugin to work (I read the code), we have to publish not under "connection" but under "connection/index". And that is a real issue. Because :
- if you have 2 lines with the same index, both actions are triggered when only one is true 
- you cannot publish different values on the same topic (e.g. "connection/1"), because of previous point. I'm using this plugin to switch ON/OFF my LEDs, and because of the index thing, I HAVE to use 2 indexes (so 2 topics) to command one device. So I send "ON" to 2 topics (index 1 and 2), and "OFF" to 2 topics, and I have 1 rule for indexes "1" and one rule for index "2" in the plugin to get what I want. That is awkward don't you think ?

Why the index thing ? Why not have the plugin exactly as the documentation suggests, screenshot included, that is to say without index ?

jneilliii, 2019-11-17:
> - if you have 2 lines with the same index, both actions are triggered when only one is true
> - you cannot publish different values on the same topic (e.g. "connection/1"), because of previous point. [...]

I don't remember why I had that index in there to be honest.  I'm sure there was some reason, but I would consider removing it if you are willing to do testing on the modified version.  I don't use the plugin personally and am looking into adding Subscribe methods within the main MQTT plugin directly for printer/job control using something like a `control` topic.

jneilliii, 2019-11-17:
I just uploaded the 0.1.1 branch that has removed the index from the equation.  Let me know if it works for you. You can install using the URL below in plugin manager.

https://github.com/jneilliii/OctoPrint-MQTTSubscribe/archive/0.1.1.zip

aschor, 2019-11-17:
Thanks, you're faster than me :)I'll be able to test it Tuesday, not before. I'll report back

https://github.com/jneilliii/OctoPrint-MQTTSubscribe/archive/0.1.1.zip

aschor, 2019-11-19:
reporting as stated : 

2019-11-19 20:47:09,500 - octoprint.plugin.core - ERROR - Error loading plugin mqttsubscribe
Traceback (most recent call last):
  File "c:\users\arnaud\devel\octoprint\src\octoprint\plugin\core.py", line 863, in _import_plugin
    instance = imp.load_module(module_name, f, filename, description)
  File "c:\users\arnaud\devel\octoprint\venv\lib\site-packages\octoprint_mqttsubscribe\__init__.py", line 67
    self.mqtt_publish("octoprint/plugins/mqttsubscribe/response/%s" % r.text
       ^
SyntaxError: invalid syntax

Lines 66,67,71,72 are missing the closing parenthesis.

After correcting this, the index is still in the user interface, I'm not sure then how to remove it, I'm not that far in the plugin devel tutorial :D

jneilliii, 2019-11-19:
Oops, didn't mean to remove those end parenthesis...I'll correct that. As for the field still being in settings, you should be able to just ignore those and still verify functionality.  If the actual subscriptions work right then I can remove them from the UI.

jneilliii, 2019-11-19:
Just updated and pushed to the same branch.  You should be able to reinstall using the URL above.

aschor, 2019-11-20:
Nope. Seems not NOT subscribe to any topic with this version. I tried publishing under 
octoprint/plugins/mqttsubscribe/printer/command/
octoprint/plugins/mqttsubscribe/printer/command/1
octoprint/plugins/mqttsubscribe/printer/command/2

nothing happens.

With only 2 lines of configuration, both on printer/command

jneilliii, 2019-11-20:
Ok, I'll do some debugging this weekend and see what's going on. It might be something as simple as a missing slash somewhere.

jneilliii, 2019-11-21:
Ok, I think I have it working now.  At least in my limited testing it is.  Here's an example config, and the responses shown in a mqtt client.

[image]

[image]

[image]

There are some other tweaks I want to do, like right now if you add a new topic that hasn't been configured before you'll need to restart octoprint after adding it in order to subscribe to the topic. I'll also remove all the index related UI stuff.

aschor, 2019-11-24:
Hi, I did a few tests, and no sorry, there is still an issue. This is my subscriptions :
[image]

because of that (I added a few logger.info lines), at startup I get : 
```
2019-11-24 14:30:02,709 - octoprint.util.pip - INFO - ==> pip ok -> yes
2019-11-24 14:30:02,721 - octoprint.plugins.mqttsubscribe - INFO - Subscribing to printer/command
2019-11-24 14:30:02,721 - octoprint.plugins.mqttsubscribe - INFO - Subscribing to printer/command
2019-11-24 14:30:02,723 - octoprint.plugins.mqttsubscribe - INFO - Subscribing to connection
```

So 2 subscriptions to the same topic. So [this](https://github.com/jneilliii/OctoPrint-MQTTSubscribe/blob/cb9e9b0afbcf76f27d857b119cd5336dd22e21c7/octoprint_mqttsubscribe/__init__.py#L53) functions is called 2 times when ONE command is received, and because of that it always fire 2 the 2 actions .

those are the logs of only ONE mqtt publication on "printer/command" topic : 

2019-11-24 14:32:19,025 - octoprint.plugins.mqttsubscribe - INFO - Received from octoprint/plugins/mqttsubscribe/printer/command|{ "command": "RESPOND TYPE=command MSG=action:relaisLeds**Off**"}
2019-11-24 14:32:19,026 - octoprint.plugins.mqttsubscribe - INFO - command match
2019-11-24 14:32:19,028 - octoprint.plugins.mqttsubscribe - INFO - POST request : http://localhost:5000/api/printer/command |data : { "command" : "RESPOND TYPE=command MSG=action:relaisLeds**On**"}

2019-11-24 14:32:19,042 - octoprint.plugins.mqttsubscribe - INFO - Received from octoprint/plugins/mqttsubscribe/printer/command|{ "command": "RESPOND TYPE=command MSG=action:relaisLeds**Off**"}
2019-11-24 14:32:19,042 - octoprint.plugins.mqttsubscribe - INFO - command match
2019-11-24 14:32:19,043 - octoprint.plugins.mqttsubscribe - INFO - POST request : http://localhost:5000/api/printer/command |data : { "command" : "RESPOND TYPE=command MSG=action:relaisLeds**Off**"}

from my point of view : 
1. issue1 : we subscribe 2 times to the same topic, with different "cmd"
2. issue2 : we POST (or GET) the "cmd" (of the subscription) calling the function "_on_mqtt_subscription" instead of the t["subscribecommand"]  of the config matching the received mqtt message

jneilliii, 2019-11-24:
1. Maybe that's why I put the index thing in there. The subscription is fired twice, but it only runs the command that matches the one sent to the subscription. Yes, there's more overhead, but it's not sending both commands to the api.
2. Don't understand the concern here. The plugin acts as an api proxy basically. The command being sent is matched against the message to the topic, so shouldn't matter which one used because `t["subscribecommand"] == message`.

jneilliii, 2019-11-24:
I just pushed updates to the 0.1.1 branch that eliminates the duplicate subscriptions to the same topic. Install using the URL below in plugin manager.  Those logging messages were already in there, you just have to go into OctoPrint's logging section and set the plugin to DEBUG.

[image]

https://github.com/jneilliii/OctoPrint-MQTTSubscribe/archive/0.1.1.zip

aschor, 2019-11-24:
>     1. s the one sent to the su

no, it actually also did run the command that was NOT matching the received message. thanks for the debuging tip.

Here is some more logging I added at the line referenced  :
`            if topic == "octoprint/plugins/mqttsubscribe/" + t["topic"]  and message == t["subscribecommand"] :
                self._logger.info('command match ; topic : '+topic+' ; message: '+message+' ; subscribecommand : '+t["subscribecommand"]+' ; cmd(at subscription) : '+ kwargs.get('cmd', 'None'))
`

and here is the result, see the non-matching "cmd(at subscription)" Vs the received message and the 2 POST requests :

`
2019-11-24 19:33:18,812 - octoprint.plugins.mqttsubscribe - INFO - Received from octoprint/plugins/mqttsubscribe/printer/command|{ "command": "RESPOND TYPE=command MSG=action:relaisLedsOff"}
2019-11-24 19:33:18,812 - octoprint.plugins.mqttsubscribe - INFO - command match ; topic : octoprint/plugins/mqttsubscribe/printer/command ; message: { "command": "RESPOND TYPE=command MSG=action:relaisLedsOff"} ; subscribecommand : { "command" : "RESPOND TYPE=command MSG=action:relaisLedsOff"} ; cmd(at subscription) : { "command" : "RESPOND TYPE=command MSG=action:relaisLedsOn"}
2019-11-24 19:33:18,813 - octoprint.plugins.mqttsubscribe - INFO - POST request : http://localhost:5000/api/printer/command |data : { "command" : "RESPOND TYPE=command MSG=action:relaisLedsOn"}

2019-11-24 19:33:18,832 - octoprint.plugins.mqttsubscribe - INFO - Received from octoprint/plugins/mqttsubscribe/printer/command|{ "command": "RESPOND TYPE=command MSG=action:relaisLedsOff"}
2019-11-24 19:33:18,834 - octoprint.plugins.mqttsubscribe - INFO - command match ; topic : octoprint/plugins/mqttsubscribe/printer/command ; message: { "command": "RESPOND TYPE=command MSG=action:relaisLedsOff"} ; subscribecommand : { "command" : "RESPOND TYPE=command MSG=action:relaisLedsOff"} ; cmd(at subscription) : { "command" : "RESPOND TYPE=command MSG=action:relaisLedsOff"}
2019-11-24 19:33:18,835 - octoprint.plugins.mqttsubscribe - INFO - POST request : http://localhost:5000/api/printer/command |data : { "command" : "RESPOND TYPE=command MSG=action:relaisLedsOff"}
`

And here his what I think is the fix to this issue (check also the kwargs) : 
`if topic == "octoprint/plugins/mqttsubscribe/" + t["topic"]  and message == t["subscribecommand"] and message == kwargs.get('cmd', 'None')  :`

which yelds this correct output, with only one POST command (but with the 2 received messages since I've not yet updated to your patch to remove double subscription) :

`2019-11-24 19:39:30,746 - octoprint.plugins.mqttsubscribe - INFO - Received from octoprint/plugins/mqttsubscribe/printer/command|{ "command": "RESPOND TYPE=command MSG=action:relaisLedsOn"}
2019-11-24 19:39:30,746 - octoprint.plugins.mqttsubscribe - INFO - command match ; topic : octoprint/plugins/mqttsubscribe/printer/command ; message: { "command": "RESPOND TYPE=command MSG=action:relaisLedsOn"} ; subscribecommand : { "command": "RESPOND TYPE=command MSG=action:relaisLedsOn"} ; cmd(at subscription) : { "command": "RESPOND TYPE=command MSG=action:relaisLedsOn"}
2019-11-24 19:39:30,747 - octoprint.plugins.mqttsubscribe - INFO - POST request : http://localhost:5000/api/printer/command |data : { "command": "RESPOND TYPE=command MSG=action:relaisLedsOn"}
2019-11-24 19:39:30,766 - octoprint.plugins.mqttsubscribe - INFO - Received from octoprint/plugins/mqttsubscribe/printer/command|{ "command": "RESPOND TYPE=command MSG=action:relaisLedsOn"}
`

anyways, I'll now checkout your new modifications regarding dual topic subscription, and see what happens, and off course report back :)

aschor, 2019-11-24:
Reporting back for latest changes : all seems OK, I get no more dual subscription, and no more dual (opposite in my case) commands sent   👍

jneilliii, 2019-11-24:
> (except that I hate tabulations ^^)

I assume you mean tabs versus spaces in code?  Aged old debate...I only use it because my editor automatically puts it in.

> I get no more dual subscription

This makes sense to me now that I think about it more.  Because of the duplicated subscription the commands were getting called for each iteration, so if you had 3 on the same topic it would send the command 3 times, etc.  Gad removing the duplicated subscriptions resolve that issue.

From your perspective (outside of tabs vs spaces) this is ready to release?

aschor, 2019-11-25:
Hi, yes, from the tests I've run, all MQTT messages seem OK from my point of view ; but there is still the U.I to update to remove the indexes.

Another point I'm not sure is the migration of config from "with index" to "no index"

jneilliii, 2019-11-25:
Thanks, I can handle the removal of index from the ui and during upgrade to eliminate any future problems. I'll do that prior to official release.

closed 2019-11-29 by jneilliii

aschor, 2019-12-14:
@jneilliii Hi, upgraded today my octoprint with release 1.1.0 from octoprint. Olny thing I did.

Sadly now, I get nothing sent to my mqtt broker, apart from "OctoPrint-MQTTSubscribe monitoring.". The events are not sent anymore.

Do you have an idea please ? I set plugin "MQTT" to debug, and it says it DOES send messages but it does not :/