jneilliii/OctoPrint-MQTTSubscribe #7: Remove matching of commands

mdjurfeldt, 2020-01-29:
Hi,

Thanks for your exciting plugin!

Maybe there is something which I haven't understood, but why do you match the payload against a commands retrieved from settings at line 71 in __init__.py:

  and message == t["subscribecommand"]

?

Why is it needed? Wouldn't it be better just to remove this matching? That would allow message payloads to have variable values, e.g., when setting tool target temperature.

Best regards,
Mikael

jneilliii, 2020-01-29:
At the time of programming this plugin it was a means in which to minimize the potential of unwanted commands being sent to the printer. By requiring the user to enter a payload to match, they would know exactly what could be done via MQTT.

> That would allow message payloads to have variable values, e.g., when setting tool target temperature.

Could you provide an example of what you mean?

mdjurfeldt, 2020-02-03:
In order to explain more clearly what I mean by variable values in messages, I made some smaller adjustments in this fork:

https://github.com/mdjurfeldt/OctoPrint-MQTTSubscribe

The basic idea is to treat the MQTT topics as one API and the REST API as a different one and *translate* between them rather than trying to mix them.

In my code, there are a couple of extra fields in Settings. To set the temperature of tool0 you could for example set:

topic=myprinter/settemp/tool0
extract=
rest=/api/printer/tool
command={ "command" : "target", "targets" : { "tool0" : %s } }

To, for example, set temperatures of multiple tools at once, you should be able to do (haven't tested this):

topic=myprinter/settemp/tools
extract=[*]
rest=/api/printer/tool
command={ "command" : "target", "targets" : { "tool0" : %s, "tool1" : %s, "tool2" : %s, "tool3" : %s } }

That should make it possible to send an MQTT message like [215.0, 230.0, 255.0, 240.0].

The extract expression here is jsonpath syntax (I used jsonpath_rw) and the principle is that each *match* of the jsonpath expression feeds one %s in command. That gives some flexibility such that we also can receive messages of the form: { temp1 : 215.0, temp2 : 230.0, temp3 : 255.0, temp4 : 240.0 } through:

extract=temp1|temp2|temp3|temp4

Another modification I did was to publish status and response as one json data object under a topic formed from topic + "/response", since I thought it more natural to group the response under the topic in question.

What do you think?

jneilliii, 2020-02-04:
Thanks sho much for this, I'm techie throughout the week, but in regards to extract. Wouldn't it be better to do some form of parameter token so it would work more like the [custom controls](https://docs.octoprint.org/en/master/features/custom_controls.html#sec-features-custom-controls) do where you can reference the same token multiple times within the command being sent? For example setting a custom command like this
```
M117 %(parameter1)s %(parameter2)s %(parameter1)s
```
would get translated to
```
M117 one two one
```
if given the parameters: parameter1 = one and parameter2 = two. 

Hope that makes sense. I'm only vaguely familiar with the custom controls after writing a new [plugin](https://github.com/jneilliii/OctoPrint-TerminalCommandsExtended) that mimics many of the features of custom controls, but puts them on the terminal tab instead.

mdjurfeldt, 2020-02-04:
The extract matching uses the very powerful jsonpath language which allows us to flexibly pick parts from a json message. Then, we compose a command currently using %s.

Along with what you suggest, instead of using %s, we could enumerate the matches. When this is done in regular expressions, the matches are named \1, \2 etc. We could, instead of using %s use %0, %1 etc (or perhaps %1, %2?). That would allow for reusing one match as you outline.

What do you think?

mdjurfeldt, 2020-02-04:
Just to clarify:

In my mind, it is better to keep the jsonpath language rather than to introduce some kind of command specification similar to custom controls (where the input parameters are declared). The reason why I think this is that jsonpath is specialized for the task of extracting data from json and we can't have the same flexibility without it. Also, note that the idea of using jsonpath for extracting data from messages is used in the same way in the Android app MQTT Dash.

However, we *can* name the matches differently and get the functionality that you request.

BTW, after having thought about this for a while, I tend to prefer the regexp names \1, \2, ... since these are familiar names for matches (and shorter than %(parameter1)s or %(match1) ).

mdjurfeldt, 2020-02-04:
Example:

topic=myprinter/lcdmessage
extract=$.param1|$.param2
rest=/api/printer/command
command=M117 \1 \2 \1

Then a message { "param1" : "one", "param2" : "two" } on the topic myprinter/lcdmessage would issue the printer gcode:

M117 one two one

(of course, "param1" etc is here completely arbitrarily chosen names)

jneilliii, 2020-02-05:
I wasn't meaning specifically to do the same method, but wasn't familiar with jsonpath.  After looking it up and seeing your example above that is definitely along the lines of what I meant. I assume that is possible currently in your fork's implementation or would it need to be modified a bit?  I can install and review the functionality and wrap my mind around the workflow. I'm more of a hands-on/visual learner.

mdjurfeldt, 2020-02-05:
I need to replace %s with \1, \2 in command substitution. Let me do that
and I'll then post a note here.

mdjurfeldt, 2020-02-05:
OK, now I think my fork does what we have said. Please give it a try.

One note, though: Apparently, there is a problem of priority among operators in the jsonpath expression above. It should be ($.param1)|($.param2).

However, in this case we could just say param1|param2

jneilliii, 2020-02-05:
If you don't mind, please re-base your fork to the 0.1.2 branch in my repo.  I have some additional changes in there to handle upgrade process that will add the new fields and clean-up previous fields no longer in use and some tweaks to the API key to make it a little easier. While doing those changes I have a quick question.  What is the logic behind adding the `rest` field since the `/api/` part was already getting created on the backend [here](https://github.com/jneilliii/OctoPrint-MQTTSubscribe/blob/5fd4e05a9a79d65de419bb8f8d6f9270e99b7582/octoprint_mqttsubscribe/__init__.py#L78)?

mdjurfeldt, 2020-02-05:
> If you don't mind, please re-base your fork to the 0.1.2 branch in my repo.

I'm unsure if it is a good idea that I make the rebase. Isn't it better that I create a pull request against your 0.1.2 branch and that you choose, for example, squash and merge?

Maybe it's just that my git/github knowledge is poor but while I *can* do the rebase in my local repository, this will rather create a merge when trying to push it to my github repository, I think. Or can you instruct me otherwise?

> While doing those changes I have a quick question. What is the logic behind adding the `rest` field since the `/api/` part was already getting created on the backend [...]

As I explained earlier, I want to *separate* the MQTT and REST APIs. Your solution, in contrast, makes hardwired ties between them by fixing an MQTT topic prefix and insisting on using the REST API suffix as topic suffix. (This is a no-no in computer science.)

Among examples demonstrating the greater flexibility gained by separating the API:s are:

* Assume that you have multiple printers but still a single MQTT broker. Then, you probably want to use different topic prefixes for the different printers, right?

* By making the tie between MQTT topic suffix and REST API pathname, you impose the structure of the REST API onto the MQTT API. This might not be what you want to do. In some cases, you might want to use a collection of different MQTT topics issuing different variations of the same REST API path, with the structure of the data portion varying. One concrete example is that you might want a specific topic for displaying something on the LCD, e.g. myprinter/display, and another topic moving the printer to a coordinate, e.g. myprinter/moveto. Both would use the REST API path /api/printer/command.

* In some setups, you might want to pick up an MQTT topic en route to some other device, i.e., a topic also destined for another device. For example, the printer might be embedded in a larger system including a controller for an enclosure. In this case, maybe both dvices want to listent to the topic mysystem/shutdown, which in the printer would result in /api/system/commands/core/shutdown.

jneilliii, 2020-02-05:
> (This is a no-no in computer science.)

Sorry, I'm a self-taught programmer so don't know these kinds of things.

> I'm unsure if it is a good idea that I make the rebase. Isn't it better that I create a pull request against your 0.1.2 branch and that you choose, for example, squash and merge?

I'm not really great with git either, but your changes are already merged into 0.1.2, just wanted to make sure that my changes made there way back to your fork to ease merging pull requests together in the future.

> Among examples demonstrating the greater flexibility gained by separating the API:s are:

Great points, that makes sense now, but I wonder if it's necessary to make the default for that field `/api/' rather than having it build the url with the `/api/' part already included and then just appending the newly created rest field into it? Similar to the way I was using the topic before, but with the rest field instead.

mdjurfeldt, 2020-02-05:
> I'm not really great with git either, but your changes are already merged into 0.1.2, just wanted to make sure that my changes made there way back to your fork to ease merging pull requests together in the future.

OK, so you do your development work in your 0.1.2? I've created a 0.1.2 in my repository now which is even with yours. In the future, I will try to create feature branches instead of working directly on a main branch.

> Great points, that makes sense now, but I wonder if it's necessary to make the default for that field `/api/' rather than having it build the url with the `/api/' part already included and then just [...]

Well, for an experienced programmer it is cleaner to not break up the REST API. Note, if you go to the [Octoprint REST documentation](http://docs.octoprint.org/en/master/api/printer.html#issue-a-tool-command), how they state the full REST path for each command. It's treated as a unit. That is cleaner and less confusing.

mdjurfeldt, 2020-02-05:
If you're worrying about the small fields, I've myself wondered if these templates allow for breaking up one topic entry into two lines? Then, the topic part and the REST part could be on different lines and there would be more space.

jneilliii, 2020-02-05:
> how they state the full REST path for each command.

Another great point, being able to copy/paste from the documentation methods makes sense. Would be nice to pull that into a drop down somehow. You don't happen to know if the REST api has a call for available options, similar to a schema request (that's what I call it at least)?

> If you're worrying about the small fields, I've myself wondered if these templates allow for breaking up one topic entry into two lines?

I can manage that easily using an additional details template/modal dialog. I've done it for several of my other plugins that have this same kind of issue, where there's not enough real estate in a single row. The Terminal Controls Extended plugin is an example. I'll work on that and get your feedback and do additional testing/workflow with the changes you made.

mdjurfeldt, 2020-02-06:
>Would be nice to pull that into a drop down somehow. You don't happen to know if the REST api has a call for available options, similar to a schema request (that's what I call it at least)?

Sorry---don't know. My guess is no. I agree that something like that would be convenient---as long as one can also type in the path---but that should then also come with a template for the data part, otherwise one would anyway have to lookup the documentation.

> I can manage that easily using an additional details template/modal dialog. I've done it for several of my other plugins that have this same kind of issue, where there's not enough real estate in a [...]

Sounds great!

Probably, one should more gracefully handle the index out of range error which happens if e.g. \2 occurs in the command when there is only one jsonpath match. For example, there could be a try clause around the matches[...] expression with an exception handler that raises an error about reference to missing match.

buzurk26, 2020-02-09:
Hi, How would you configure this new dev branch to handle a "Connect" requests, to tell OctoPrint to connect to the printer.

jneilliii, 2020-02-09:
Haven't been able to test, but I think setting the item to
```
topic=/myprinter/connect  (whatever topic you want to monitor)
extract=
rest=/api/connection
type: post
command={"command": "connect"}
```
and publish to 
```
/myprinter/connect
```
would work,

buzurk26, 2020-02-09:
ill give it a go now
Im not seeing it come up in MQTT explorer though

buzurk26, 2020-02-09:
Trying from MQTT Broker direct in homeassistant also

https://imgur.com/dvWsWtm

buzurk26, 2020-02-09:
**UPDATE:** i thnk we are getting a touch closer

If i publish this https://imgur.com/CCLHMl5

I get this https://imgur.com/i935ksL  in the OctoPrint Web UI, but nothing happens

jneilliii, 2020-02-09:
I'll have to defer to @mdjurfeldt here until I've had a chance to test everything. The command is obviously getting to octoprint now, so the next step is verifying the command is getting posted to the rest api correctly.  Any errors in your logs?

mdjurfeldt, 2020-02-09:
Hi @buzurk26,

I tested the setting that @jneilliii suggested above on a clean install of his branch 0.1.2 and it works for me. (But note that there should be no initial "/" in MQTT topics since that internally causes an unnecessary empty string name component.)

But your Web UI response indicates a misunderstanding: You have sent { "command": "connect" } as message payload while what @jneilliii meant above was to enter it in the command field in the OctoPrint-MQTTSubscribe settings.

jneilliii, 2020-02-09:
So I just tested and there are some issues after initially installing as @mdjurfeldt pointed out in issue #10. While testing I also noticed errors on save in ticket #11. Once you've added one topic in correctly restart OctoPrint for it to subscribe correctly to the configured topic.

Your topic configuration should be like this.

```
topic=myprinter/connect  (whatever topic you want to monitor)
extract=
rest=/api/connection
type: post
command={"command": "connect"}
```

> But your Web UI response indicates a misunderstanding: You have sent { "command": "connect" } as message payload while what @jneilliii meant above was to enter it in the command field in the OctoPrint-MQTTSubscribe settings.

In it's current state with 0..1.2 branch if you don't publish a json object as payload you get a json load error. I was testing by just publishing to the topic without data and it choked. If I send the same message as configured in the plugin's settings everything works perfectly.

[image]

mdjurfeldt, 2020-02-09:
Just submitted pull request #13 which should allow for empty MQTT messages.

jneilliii, 2020-02-09:
@mdjurfeldt, quick question on your on_settings_save changes.  It seems it's causing some type of error on save and not working as expected.
```
2020-02-09 15:39:41,528 - octoprint.plugins.mqttsubscribe - DEBUG - Unsubscring from topics: ['myprinter/connection']
2020-02-09 15:39:41,532 - octoprint.server.api.settings - WARNING - Could not save settings for plugin MQTT Subscribe (0.1.2) since it called super(...)
2020-02-09 15:39:41,535 - octoprint.server.api.settings - WARNING - in a way which has issues due to OctoPrint's dynamic reloading after plugin operations.
2020-02-09 15:39:41,536 - octoprint.server.api.settings - WARNING - Please contact the plugin's author and ask to update the plugin to use a direct call like
2020-02-09 15:39:41,539 - octoprint.server.api.settings - WARNING - octoprint.plugin.SettingsPlugin.on_settings_save(self, data) instead.
```

Are you seeing the same thing on your side? The odd part is that the command it says to use is being used already, so I'm confused.

jneilliii, 2020-02-09:
I figured it out.  Will be pushing some changes to the 0.1.2 branch shortly for the improved UI and the other issues we've found so far.

jneilliii, 2020-02-09:
I just pushed updates to the dev branch with the improved UI and the bug fixes included. So far with my testing I haven't been able to reproduce any errors and it's working as expected now.

buzurk26, 2020-02-10:
Great, thank you so much

How do i re-pull this, should i uninstall and reinstall, or just add plugin
again?

jneilliii, 2020-02-10:
You should be able to just re-use the same url from before for the 0.1.2 branch in plugin manager (before for convenience). Make sure if not prompted to restart and reload (shift+f5 or ctrl+f5 depending on browser) the web interface that you do that manually. 

https://github.com/jneilliii/OctoPrint-MQTTSubscribe/archive/0.1.2.zip

buzurk26, 2020-02-10:
Will it just over-write the current install 0.1.2?
Or should i remove that plugin first?

jneilliii, 2020-02-10:
Yes, it will just overwrite. Settings will be transferred.

jneilliii, 2020-02-12:
@mdjurfeldt, just ran into an issue while doing some final testing/prep for release.  In the on_settings_save, if you have two configured topics that are the same there is an error and the topic ends up being unsubscribed. Here's an example debug excerpt with the current 0.1.12 branch.

Initial startup subscriptions.
```
2020-02-11 22:39:41,165 - octoprint.plugins.mqttsubscribe - DEBUG - Subscribing to octoprint/plugins/mqttsubscribe/connection
2020-02-11 22:39:41,170 - octoprint.plugins.mqttsubscribe - DEBUG - Subscribing to printer/connect
```
On Save
```
2020-02-11 22:41:27,832 - octoprint.plugins.mqttsubscribe - DEBUG - Subscribing to printer/connection
2020-02-11 22:41:27,835 - octoprint.plugins.mqttsubscribe - DEBUG - Unsubscribing from octoprint/plugins/mqttsubscribe/connection
2020-02-11 22:42:07,220 - octoprint.plugins.mqttsubscribe - DEBUG - Exception: list.remove(x): x not in list
```

My thought process here is that you may want to have multiple REST API calls with the same MQTT topic as a form of batch action. Is there some logic that I may be missing here? Any ideas on how to fix other than restricting the topics as being unique?

jneilliii, 2020-02-12:
I think I got it.  Added a check to make sure the topic was in the to_unsubscribe list prior to trying to remove it.

mdjurfeldt, 2020-02-16:
Nice! (You could use an and-expression instead of two if:s, but that's a detail.)

PR #14 (2020-05-01): 0.1.2
Disassociate MQTT topics from REST API, resolves #7, resolves #15 
Added jsonpath parameter replacement of messages using extract method, see #7 for details
Settings UI overhaul with easier api key generation, resolves #10, resolves #11 
Option to disable popup messages per topic, resolves #6
Bug fixes, resolves #12 
REST API response now publishes to `<topic>/response/`
closed 2020-05-01 by jneilliii