jneilliii/OctoPrint-MQTTSubscribe #11: Error on Save

jneilliii, 2020-02-09:
```
2020-02-09 12:38:04,852 - octoprint.server.api.settings - ERROR - Could not save settings for plugin MQTT Subscribe (0.1.2)
Traceback (most recent call last):
  File "c:\octoprint\venv\lib\site-packages\octoprint\server\api\settings.py", line 563, in _saveSettings
    plugin.on_settings_save(data["plugins"][plugin_id])
  File "c:\octoprint\venv\lib\site-packages\octoprint_mqttsubscribe\__init__.py", line 61, in on_settings_save
    self.mqtt_unsubscribe (self._on_mqtt_subscription, topic)
AttributeError: 'MQTTSubscribePlugin' object has no attribute 'mqtt_unsubscribe'
```

Need to check to see if topic is subscribed or not before trying to unsubscribe.

mdjurfeldt, 2020-02-09:
> Need to check to see if topic is subscribed or not before trying to unsubscribe.

Hi, maybe you have already solved this also, but the cause is not what you suggest above. This code is actually precisely doing that: It (at least tries to) unsubscribe topics that that *are* subscribed but that are no longer used.

The direct cause of the error is a bit strange: Python complains about self.mqtt_unsubscribe not being initialized, which it should have been in on_after_startup:

    if "mqtt_unsubscribe" in helpers:
    	self.mqtt_unsubscribe = helpers["mqtt_unsubscribe"]

jneilliii, 2020-02-09:
I think your other unsubscribe stuff actually resolves this issue.  I haven't been able to reproduce it yet.

PR #14 (2020-05-01): 0.1.2
Disassociate MQTT topics from REST API, resolves #7, resolves #15 
Added jsonpath parameter replacement of messages using extract method, see #7 for details
Settings UI overhaul with easier api key generation, resolves #10, resolves #11 
Option to disable popup messages per topic, resolves #6
Bug fixes, resolves #12 
REST API response now publishes to `<topic>/response/`
closed 2020-05-01 by jneilliii