jneilliii/OctoPrint-MQTTSubscribe #19: receiving No JSON object could be decoded

dannieboiz, 2020-08-05:
I'm using homeseer to try to automatically connect the printer upon power on. In the event, the topic is "octoprint/ender3" the message/payload is "connect" 

Seems the message is being received by octoprint but it's throwing an error "No JSON object could be decoded"

jneilliii, 2020-08-05:
Your message payload should match the requirements of the REST API that you are calling. The specific task you are trying to achieve is documented on the readme actually, in the screenshot.

[image]

dannieboiz, 2020-08-05:
That solves my problem. THANKS YOU!!! It can no autoconnect.

closed 2020-08-05 by dannieboiz