jneilliii/OctoPrint-MQTTSubscribe #20: JSON Substitution escapment

fsironman, 2020-08-13:
Hello I was testing this Plugin and came across a problem when trying to concatenate string parameter together.

```
topic=printer/ender3/fan/speed
JSONPath Extract= ($.fanspeed)|($.cmd2)
REST API=/api/printer/command
REST Parameters=   {"commands": [  "M106 S\1", "\2" ]}
```

Sending the following message over mqtt:
```
{
	"fanspeed": "64",
	"cmd2": "QUERY_ENDSTOPS"
}
```

Result in the following API post DATA going out:
2020-08-13 18:01:42,260 - octoprint.plugins.mqttsubscribe - DEBUG - DATA: 
```
  {"commands": [
    "M106 S"64"",
    ""QUERY_ENDSTOPS""
  ]}
```
Returning me the Error: 
Sent message: printer/ender3/fan/speed - { "status" : 400, "response" : Malformed JSON body in request },

I can "fix" it by sending 

```
{
	"fanspeed": 64,
	"cmd2": "QUERY_ENDSTOPS"
}
```
and using REST Parameters=   {"commands": [  "M106 S\1", \2 ]}

But then I lose the ability to do something like:
```
{
	"fanspeed": 64,
	"displaymsg": "Fanspeed set to 25%"
}
```

using REST Parameters=   {"commands": [  "M106 S\1", "M117  \2" ]}

Resulting in invalid JSON data for the API call:
2020-08-13 18:16:32,931 - octoprint.plugins.mqttsubscribe - DEBUG - DATA: 
```
  {"commands": [
    "M106 S64",
    "M117 "Fanspeed set to 25%""
  ]}

```

tldr: JSON Substitution should handle strings without "" so you can do concatenation with strings

jneilliii, 2020-08-13:
@mdjurfeldt any suggestions on how to fix this?

jneilliii, 2020-08-29:
Why not do something like this?

topic=printer/ender3/fan/speed
JSONPath Extract= ($.cmd1)|($.cmd2)
REST API=/api/printer/command
REST Parameters= {"commands": [  \1, \2 ]}

and send the json message below to the topic...

```
{
	"cmd1": "M106 S64",
	"cmd2": "M117 Fanspeed set to 25%"
}
```

jneilliii, 2020-08-29:
I'm also curious if possibly putting the slash references outside of the json " would also resolve this?

```
{"commands": [ "M106 S"\1, "M117 "\2 ]}
```

jneilliii, 2020-08-29:
Yeah, the last one doesn't work at all. It has the same issue,

```
{"commands": [ "M106 S""64", "M117 ""Fanspeed set to 25%" ]}
```

jneilliii, 2020-08-29:
So I think I have a possible solution to this issue, but will require a change of your command format to work properly and I need to do some additional testing. Rather than using the `\#` replacement approach, I switched it to the python format approach, which is much more flexible. Some simple examples can be found [here](https://www.w3schools.com/python/ref_string_format.asp) that you can play with to get the string correct. The replacement is still index based from the JSONPath extraction, but starts at 0. So in your above example you would use something like this.

```
{"commands": [ "M106 S{0}", "M117 {1}" ]}
```

[image]

sent message to MQTT topic

```
{"cmd1": "64","cmd2": "Fanspeed set to 25%"}
```

Result....

[image]

If you want to try it out, feel free to install my current dev version by copying/pasting the URL below into Plugin Manager > Get Mote > ...from URL

```
https://github.com/jneilliii/OctoPrint-MQTTSubscribe/archive/0.1.5.zip
```

mdjurfeldt, 2020-08-29:
Hi @jneilliii,

Sorry for late reply to this issue.

Right, there was a fundamental limitation in the old (\#) formatting scheme in that the matches \1, \2 always represented whole json objects (which in a sense is intuitive). So, splicing a string into a larger string was not supported.

This could of course be fixed by introducing new syntax, for example.

But now you seem to be on the track of supporting Python formatting. I actually thought about that option when thinking about what kind of match syntax to use, but I abandoned that idea since I thought that the resulting mixture of Python matching syntax and json syntax (which also uses "{") could introduce complex issues, especially when the formatting string contains hierarchically nested json objects. E.g., can you really handle things like { "foo": { "bar" : 17 } } ? I think you need to give this mix of syntax a deep thought. :)

Best regards,
Mikael

jneilliii, 2020-08-29:
> can you really handle things like { "foo": { "bar" : 17 } } ?

Good point, I need to verify this nested object syntax. So far my testing was only limited to a nested list in the above examples. I guess the question would be does any of the existing OctoPrint API have that nested object situation. If not, it's a non-issue really. I'm already wrapping a json object payload with additional curly brackets to resolve it being recognized as a named replacement, It would be possible to do additional regex replacements to do the same on nested curly brackets that are not the replacemet indexes, but it definitely gets more complicated. I'm going to look through the API first to determine if it's even necessary.

jneilliii, 2020-08-29:
So it does look like there are a couple of API commands in OctoPrint that support embedded objects, one example below. 

```
POST /api/printer/tool HTTP/1.1
Host: example.com
Content-Type: application/json
X-Api-Key: abcdef...

{
  "command": "offset",
  "offsets": {
    "tool0": 10,
    "tool1": -5
  }
}
```

I attempted using the double curly bracket approach in my MQTTSubscribe topic setting like below, but that doesn't work, so I wonder if there is another method of escaping embedded curly brackets inside the python format call.

[image]

jneilliii, 2020-08-29:
Ok, I take that back. This approach does work, so I will work on getting the non replacement curly brackets doubled up for proper escaping. That should work I believe long term.

jneilliii, 2020-08-29:
Alright, so I have the changes pushed now to the 0.1.5 link above. If you uninstall and reinstall using that URL and update your REST Parameters for using python replacement nomenclature with indexes of matches of the JSONPath Extraction it works with embedded objects now as well. So the example for offsets above is working now with the latest update.

mdjurfeldt, 2020-08-29:
Well, this is how people at Microsoft think. Like "people are never going to have a need for filenames longer than eight characters" or "people will never use hard drives larger than X". The proper way to write software is not to consider what seems to work at the moment but to write code that solidly works in all cases no matter what. In this case, I would advise to think about the two syntaxes you want to mix and figure out exactly where they collide and how that should be handled. Even if Octoprint doesn't have hierarchical structures right now, it might have in the future.

Best regards,
Mikael

jneilliii, 2020-08-29:
Right, and my latest changes do account for all embedded objects, so we should be good.

closed 2020-10-16 by jneilliii

DivanX10, 2022-12-24:
> So I think I have a possible solution to this issue, but will require a change of your command format to work properly and I need to do some additional testing. Rather than using the `\#` [...]

Could you show me how to correctly send a command in the format G90 G1 X50 Y50 Z30 F3000 or G91 0 200 F 3000G 90? After I send the command, the printer starts signaling and the coordinates get lost, only restarting the printer helps

https://github.com/jneilliii/OctoPrint-MQTTSubscribe/issues/37