jneilliii/OctoPrint-MQTTSubscribe #25: Single '}' encountered in format string

AJErazzor, 2020-10-22:
after an upgrade I now get the error
Single '}' encountered in format string
no other settings changed between versions

jneilliii, 2020-10-22:
can you provide the full error log so I can see what line that is happening at? Also, please share a screenshot of your settings.

AJErazzor, 2020-10-22:
I hope it’s ok to ask where I get the log from 

Thanks
Andrew

jneilliii, 2020-10-22:
Of course, go to OctoPrint's settings and in the logging section download the file octoprint.log. You can drag it into a comment here on github to upload and share. Feel free to just look in that file (text based) for the specific error potentially. Just need to get the full error message to be able to find the line that's choking up. It would also be helpful to see your settings you've configured in the plugin.

AJErazzor, 2020-10-24:
I cant see the error in there but I forced the error to appear and can see that gcode loading
[octoprint small.log](https://github.com/jneilliii/OctoPrint-MQTTSubscribe/files/5432213/octoprint.small.log)

AJErazzor, 2020-10-24:
[image]
[image]

jneilliii, 2020-10-24:
Can you try changing the REST Parameters for the connection to `{"command":{0}}`

AJErazzor, 2020-10-25:
Sorry, didnt make a difference.  I have confirmed it only happens on the connection command though

jneilliii, 2020-10-25:
Can you please share the full settings of that command, so I can see how you have the inputs configured?

AJErazzor, 2020-10-26:
Calling from openhab (home automation)

    var url = "http://<octoprintIP>/api/files/local/" + OctoprintFileUploadPath.state + "?apikey=<apikey>"
    var contenttype = "application/json"
    var POSTrequest = '{"command":"select"}'
    var output = sendHttpPostRequest(url, contenttype, POSTrequest)

AJErazzor, 2020-10-26:
I am sending from prusa slicer (2.3 alpha 2) is this what you mean?

jneilliii, 2020-10-26:
No I mean can you share a screenshot of that one connection command by pressing the edit button and showing the pop-up editor. I want to try and re-create exactly what you have configured.

jneilliii, 2020-10-26:
Ok, I figured out what's going on. Due to the switch to python formatting, the json object being sent to the api currently has to have some whitespace in it. So if you update your connect topic to this it should work again.

[image]

Notice the space after the opening `{` and before the closing '}'. So it should be `{ "command":{0} }` until I have time to figure out a way to fix this.

AJErazzor, 2020-10-27:
OK - confirmed the above removes the error, however, the command itself doesnt fire
[image]

Log shows:
2020-10-27 11:26:00,838 - tornado.access - WARNING - 400 POST /api/connection (::1) 18.12ms

any ideas?

jneilliii, 2020-10-27:
can you share a screenshot of your setting in the plugin and what you're sending to the topic as a message? For the example I gave above I just send the word `connect` or `disconnect` and it will handle both.

AJErazzor, 2020-10-27:
[image]
[image]
this is from Openhab where I have an on/off switch

AJErazzor, 2020-10-27:
OK I got it working if I remove the JSONPATH and just send connect or disconnect - I think issue 7 talks about the /1 syntax that I was using before - perhaps that has been deprecated in this release?
Turns out I didnt need it anyway but not sure if others are relying on this functionality?

jneilliii, 2020-10-27:
> I think issue 7 talks about the /1 syntax that I was using before - perhaps that has been deprecated in this release?

Yeah, that was one of the differences made in this latest release, switching, in preference of python string formatting nomenclature `{0}`.  That being said what you had before should have worked with this new format with the $.usbconnect path in place because you were sending that json message formatted that way. I'll do some more debugging as I work out the space issue and see why that wasn't working as expected.

jneilliii, 2020-10-27:
The meat of the change was discussed in #22 with a couple of different detailed examples.

AJErazzor, 2020-10-27:
OK no probs thanks

closed 2020-11-21 by jneilliii