jneilliii/OctoPrint-MQTTSubscribe #27: Turning lights on/off not working

thermoscookies, 2020-12-02:
I have MQTT and MQTT Subscribe setup and appears to be configured correctly.

Here is my command:
[image]

I am able to make the mqtt call successfully and the mqtt command makes the api call as show in the log:
`2020-12-02 19:12:21,298 - tornado.access - INFO - 200 POST /api/printer/command (127.0.0.1) 21.82ms`

but the lights don't turn on or off. Could there be an issue because of the space between the M42 command and the S255 value? I'm not sure what to check next or where to look.

jneilliii, 2020-12-02:
I think you might be running into a formatting issue with the json string and how the plugin reads it. I haven't had a chance to try and fix it yet, but if you make your rest paramaters like this I think it will work.

```
{ "command": "M42 S255" }
```

thermoscookies, 2020-12-02:
> I think you might be running into a formatting issue with the json string and how the plugin reads it. I haven't had a chance to try and fix it yet, but if you make your rest paramaters like this I think it will work.
> 
> ```
> { "command": "M42 S255" }
> ```

That fixed it, thank you for your help!

jneilliii, 2022-11-05:
Newer version should allow for the formatting issues in json and not require the whitespace anymore.

closed 2022-11-05 by jneilliii