jneilliii/OctoPrint-MQTTSubscribe #28: 2 subscriptions on the same topic & different parameters : triggers all actions

aschor, 2021-01-03:
Hi,

I have to raise again an old issue I already had raised before : https://github.com/jneilliii/OctoPrint-MQTTSubscribe/issues/4#issuecomment-557891600

I did upgrade octoprint, mqtt and mqtt subscribe. And now, my subscriptions don't work anymore. Or to be exact, the bug has re-appeared. I fire one and only one commant through mqtt, and 2 actions match and trigger opposite actions :

[image]

and this is what happens : 
[image]
[image]

with those subscriptions :
[image]

could you please investigate ? this is breaking my setup right now, I cannot swith my leds on or off ... and I had a third command subscribed to /api/printer/command, say "power off printer" ......... then I guess that switching the lights would poweroff the printer ............... :/ 

I have MQTT Subscribe (0.1.6) installed

jneilliii, 2021-01-03:
Because you're using the same mqtt topic path I think you would benefit from the jquery extraction options. Then you send the action in your payload and only have the single subscription.

aschor, 2021-01-03:
hi @jneilliii 

I'm not sure I understand you. I use the same topic path (octoprint/plugins/mqttsubscribe/printer/command) because, as I understood with previous versions, it is (was ??) the only way to talk to mqttsubscribe plugin, am I wrong here ?

Sure I would benefit from jquery extraction options, I'll look into it thanks. So, as I understand, the logic behind the pluggin has changed somehow.

jneilliii, 2021-01-03:
The if logic was changed in version 0.1.2 if I remember correctly when the jsonpath expression extraction was introduced.

aschor, 2021-01-04:
hummm OK.

I read the doc behind the new logic/philosophy, and also the jsonpath extract examples .......... and I couldn't get mything to work with the provided examples........ I always get a "Json malformed" with something like { "command": "my command  Leds\1" } in the rest parameters . Because of the "\1" being INSIDE the string

BUT. Reading the code I foud a hack that works :

[image]

Yes, the index here starts at 0 or I get a "tuple index out of range" error.... that is weird, but that works with this simple MQTT message :  { "commande" : "OFF" }
and Jsonextract : $.(commande)

(this is instructing klipper, to command Octoprint action commmand plugins)

do we close this issue, or does it stay open, if you want to implement a working "\1 \2" syntax inside strings in rest parameters ?

jneilliii, 2021-01-04:
My bad, I forgot to mention after the jsonpath extraction was added the replacements got switched over to use python token formatting instead of the \1 \2 formatting, so what you figured out is the proper approach. I will leave this ticket open to remind me to update the docs to make this a little more clear. Sorry again for all the headaches.

Arquiteto, 2021-02-20:
Okay, so what is the correct way to parse commands now? I'm trying to create MQTT switch in HA. Unfortuantely HA doesn't support sending different topics to 1 switch, so I need to use payload with for example "connect" and "disconnect" commands, and I have no idea how to make it work with parsing JSON.

jneilliii, 2021-02-20:
If I'm not mistaken you should be able to use a REST parameters below and then your payload is either `connect` or `disconnect`.
```
{ "command":"{0}" }
```

Arquiteto, 2021-02-21:
@jneilliii 

I think I'm doing this right, but unfortunatelly it doesn't work.

Here's my MQTT Subscribe config
[image]

And my MQTT message:
[image]

jneilliii, 2021-02-21:
change your payload to raw and send just the text `connect` or `disconnect` and change your JSONPath extract to be blank.

Arquiteto, 2021-02-23:
This doesn't work as well unfortunatelly :(

jneilliii, 2021-02-23:
I have confirmed that this works with just sending a payload of `connect` or `disconnect` to the configured topic. Once you've made the change, restart OctoPrint just to rule out any possible issues with subscriptions.

[image]

Arquiteto, 2021-02-23:
Okay, thank you very much for help. The problem was with the quotation marks around {0}. Without them it works.

closed 2021-02-23 by jneilliii