jneilliii/OctoPrint-MQTTSubscribe #29: tornado.access - WARNING - 403 GET /api/connection (127.0.0.1) 10.40ms

grosj, 2021-01-04:
Hello,

I'm using OctoPrint in a docker container with this plugin.

Now I'm sending a MQTT-Message
`printer/connect
{"command":"connect"}
`
or 

`printer/connect
connect
`

and both I get the following response:
`printer/connect/response
{ "status" : 403, "response" : <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>403 Forbidden</title>
<h1>Forbidden</h1>
<p>You don't have the permission to access the requested resource. It is either read-protected or not readable by the server.</p>
 }`

In the log-file I found this entry:
`2021-01-04 19:04:49,442 - octoprint.plugins.mqttsubscribe - DEBUG - Received from printer/connect|b'{"command":"connect"}'
2021-01-04 19:04:49,444 - octoprint.plugins.mqttsubscribe - DEBUG - Found match printer/connect
2021-01-04 19:04:49,496 - tornado.access - WARNING - 403 GET /api/connection (127.0.0.1) 10.40ms
`

This is my configuration:
[image]

I'm confused what's wrong!

Best Regards,
Marco

jneilliii, 2021-01-04:
403 forbidden make me think you didn't go through the process of setting up your API key in the plugin's settings?

[image]

Press the `+` sign there and then allow on the pop up that shows, and then save your settings.

grosj, 2021-01-04:
Thank you for your fast answer.
I have setting up an API key.
Now I get the following response
`{ "status" : 200, "response" : ...}`
But the Web-Frontend show the status the printer is not connected and "offline".

When I call the API per REST the connection will established and in the Web-Frontend shows online/ready.

jneilliii, 2021-01-04:
oh, small little formatting issue with your REST parameters until I have time to try and fix it. Make the REST parameter as follows and it should work. The plugin needs whitespace between the curly brackets and quotation marks. Once you change that it should work as expected I think.

```
{ "command": "connect" }
```

grosj, 2021-01-05:
Thank you for the answer.
I have change the configuration, but it don't work ...
[image]

jneilliii, 2021-01-05:
you need to use a `post` type for sending commands and performing actions.

jneilliii, 2021-01-05:
notice how here in the docs, the front of the part highlighted in blue says POST.

https://docs.octoprint.org/en/master/api/connection.html#issue-a-connection-command

grosj, 2021-01-05:
Ohh.... that's it... it runs. I don't see this little thing.

Thank you for your fast support. And sorry for my blindness.

closed 2021-01-06 by jneilliii

ryancasler, 2021-01-26:
WHITE SPACE BETWEEN THE CURLY BRACKET AND THE COMMAND?!?!?!   That's all I've been doing wrong?!?  I've been trying to figure this out forever!!!!  A little tip on the main read-me might be nice.  _grumble grumble_ 
🤡 🤪   In all seriousness though, it would be kinda helpful.  Thanks so much for the wonderful plugin!!!   👍

jneilliii, 2021-01-26:
Yeah, I keep forgetting to put that on the readme, I apologize. Documentation always falls short.

jneilliii, 2021-01-26:
Hopefully the above commit will help the next guy. Sorry for the headaches @ryancasler