jneilliii/OctoPrint-MQTTSubscribe #30: Help with shutdown

chrispi68, 2021-02-09:
Hi,

I try to shutdown OctoPrint (RasPi) via MQTT. I set up a topic and OctoPrint seems to receive the MQTT message and shows the PopUp message. But it does not shutdown. I use these parameters:

REST API: /api/system/commands/core
REST Parameters: { "command": "shutdown" }

I also tried "/api/system/commands", "/api/system/" but it did not work. The user for the API key has "SYSTEM" rights.

Can someone point me to the right direction?

Best regards
Christoph

jneilliii, 2021-02-09:
I think your API endpoint is wrong. I think it should be `/api/system/commands/core/shutdown` and your REST parameters would need to be `{ "source":"core", "action":"shutdown" }`.

chrispi68, 2021-02-11:
Thank you, this worked!

closed 2021-02-11 by chrispi68