jneilliii/OctoPrint-MQTTSubscribe #31: Q: How to capture a specific topic/message combo

aderusha, 2021-11-14:
I'm hoping to utilize this plugin to capture MQTT traffic from Tasmota on a device which controls the power to my printer.  Goal would be to catch the command for power on, then to issue an API call to octoprint to start the connection process.  I don't really grok JSONpath very well and am not having a lot of luck getting this to go.

Here are two different commands that are both received when the Tasmota device powers on the specific port I'm interested in

Message 1 Topic: `stat/power07/RESULT` Payload: `{"POWER2":"ON"}`
Message 2 Topic: `stat/power07/POWER2` Payload: `ON`

Both are sent with every power command, so catching either of them should work.

Here's what I have currently that isn't working, and it's probably because I'm doing something dumb:

Topic: `stat/power07/RESULT`
JSONPath Extract: `$.[?($.POWER2=='ON')]`
Type: `Post`
REST API: `/api/connection`
REST Parameters: `{ "command": "connect" }`

jneilliii, 2021-11-14:
Why not just use my tasmota or tasmota-mqtt plugin?

aderusha, 2021-11-15:
Because I'm a dummy and never thought to do that :D 

Thanks for the hint and all the rad stuff you make and give out to the rest of us for free, I really appreciate it!

aderusha, 2021-11-16:
Confirmed I can use that Tasmota plugin to control power on the printer.  Still having problems with connecting on power-on, but I should probably take this conversation over there.  Thanks @jneilliii!

aderusha, 2021-11-16:
OK re-opening this issue after further investigation: it appears that PSU Control does not offer a means to trigger the connection process when it detects the PSU has powered on.  The printers I'm using are all Prusa which have USB-to-serial chipsets that are powered via USB, so PortList won't work because the USB port remains online even if the printer is powered off.

So... I'm back to attempting to catch an MQTT message and send an API request into OctoPrint as a result.  Does this plugin work with a single topic that might have different messages, and just catching a specific message?  Meaning, can I catch an "ON" message and send the "connect" command (but don't do that when the "OFF" message is published).

jneilliii, 2021-11-16:
ok, when I get back around to working on this plugin I'll investigate adding the auto-connect feature when the status changes to on, while triggered from an outside source. Based on the discussion in Discord, I think you got this worked out with another fork of PSU Control, but will still investigate adding this on this end as well.

jneilliii, 2021-11-16:
Actually, I was thinking you were attempting this on the tasmota-mqtt plugin...what you want is this I think. 

https://github.com/jneilliii/OctoPrint-MQTTSubscribe/issues/29

jneilliii, 2021-11-16:
only issue is that this plugin doesn't handle if type scenarios with the json extraction stuff, so I would still recommend using one of my tasmota plugins and request the auto-connect to be integrated on external status change.

aderusha, 2021-11-17:
Roger that, that answers the question directly.  I'll go ahead and close this as I think you've sorted out the direct question posed.

Maybe a feature request: allow the subscribe function to match against the message (regexp or whatever), as the extract capability leaves some gaps in functionality.  As you noted, it only really works if the data you want to send to octopi appears directly in the MQTT message being published.  A good option would be to provide a match field, if the message matches some expression, then take the desired action.

closed 2021-11-17 by aderusha