jneilliii/OctoPrint-MQTTSubscribe #32: Invalid JSON when response is 204 NO CONTENT

morakles, 2022-03-08:
Hi, When HTTP response is 204 NO CONTENT plugin produces invalid JSON. In my case I am using this plugin to trigger relay using USB Relay plugin. My example OFF request:
[image]
USB Relay plugin return empty response body so MQTT subscribe producess:
{ "status" : 204, "response" :  "data" : {"command": "turnUSBRelayOff", "id":"1"} }
but should produce:
{ "status" : 204, "response" :  "", "data" : {"command": "turnUSBRelayOff", "id":"1"} }

It is not a big issue but I want to subscribe response topic to subscribe another service which returns relay statuses.

jneilliii, 2022-03-08:
Interesting dilemma when the API endpoint doesn't return anything. I might be able to take a look, but might also be a good idea for that plugin to return a response?

morakles, 2022-03-08:
> Interesting dilemma when the API endpoint doesn't return anything. I might be able to take a look, but might also be a good idea for that plugin to return a response?

My first thought was to raise an issue in USB relay repo but HTTP 204 code means:
_"The 204 response MUST NOT include a message-body and thus is always terminated by the first empty line after the header fields."_

jneilliii, 2022-03-08:
Yeah, makes sense. I'm not an API programmer typically so wouldn't know all the different response codes and their meanings. Will attempt a fix for this when I have the time.

jneilliii, 2022-11-05:
I think I may have figured out the cause of this. There was an issue in my response for post messages in general and was missing the delimiting comma in the JSON object.

closed 2022-11-05 by jneilliii