jneilliii/OctoPrint-MQTTSubscribe #35: List of Topics and Setting Bed Temp

BlueWings172, 2022-05-17:
Hello

Thanks for the great plugin. I just need help with the below:

1- From where I can get the list of topics that I can publish to ?

2- I'm trying to set the bed temp through MQTT so I entered the below in the editor but it not working. I does work for static temperature only:

[image]

Thanks

jneilliii, 2022-05-17:
I'll have to experiment for '#2', but '#1' is any topic you want to use, that's just how mqtt works.

BlueWings172, 2022-05-17:
Thanks for the answer. I'm new to MQTT and learning. 

I figured out the solution to the 2nd issue. I should've removed the double quotes surrounding the variable and when I changed the line to:  `{"command": "target","target": {0}}`  , it worked. Im posing the correct config below incase someone else is trying to do the same thing:

[image]

closed 2022-05-17 by BlueWings172