jneilliii/OctoPrint-MQTTSubscribe #36: Base MQTT topic should be pulled from MQTT plugin setting

tideline3d, 2022-11-05:
There's a message or two that seem to be hardcoded to publish to octoprint/ topic.   However in a multi-printer environment this will get weird.

MQTT allows you to set the base topic per device.  It'd be good if MQTTSubscribe were respect that setting.  

[image]

jneilliii, 2022-11-05:
yeah, that makes sense. It is only for that one message if I remember correctly on startup [here](https://github.com/jneilliii/OctoPrint-MQTTSubscribe/blob/80040a94f044834d3aa89e9ae3c81e188d47935d/octoprint_mqttsubscribe/__init__.py#L100).

jneilliii, 2022-11-05:
switched to using baseTopic from MQTT plugin's settings with above commit.

closed 2022-11-05 by jneilliii

tideline3d, 2022-11-05:
Verified working