jneilliii/OctoPrint-MQTTSubscribe #37: How do I send the G90 G1 X50 Y50 Z30 F3000 command to the Home Assistant using the MQTT service?

DivanX10, 2022-12-06:
Installed OctoPrint 1.8.6
Плагин: MQTT Subscribe (0.1.8)

If I send the command G90 G1 X50 Y50 Z30 F3000 through the terminal, then the printer works. How to configure it correctly so that I can send such a command via Home Assistant using the MQTT service?

[image]

If I send a command to the Home Assistant via MQTT, then it doesn't work
[image]
[image]

I tried it as they wrote [here](https://github.com/jneilliii/OctoPrint-MQTTSubscribe/issues/21#issuecomment-678969385), also does not work

[image]
[image]
[image]

Marked with a green check mark, it works
[image]

For example, roll out the platform to me with the command { "command": "G1 Y200 F3000" }
[image]

jneilliii, 2022-12-06:
well...in this screenshot
[image]
the json extract path parameter doesn't match the payload json you're sending via MQTT. `$.mycommand` should be `$.command` based on the MQTT screenshot that follows as you're sending `{ "command" : "G90..." }`.

DivanX10, 2022-12-21:
The question is relevant. There is no answer how to send a command correctly so that the printer works it correctly

DivanX10, 2022-12-24:
How can I send several commands so that the printer does not signal after sending the command and the home parking does not stop working? The coordinates of the printer are lost and only complete shutdown and turning on of the printer restore the coordinates and the printer can park.

How to specify commands correctly? { "command": "G91 G0 Y200 F3000 G90" }
G91 (transition to relative coordinates)
G0 Y200 (move by Y)
G90 (return to absolute coordinates)

DivanX10, 2022-12-24:
I found a curious thing. If I use this plugin [OctoPrint-HomeAssistant](https://github.com/cmroche/OctoPrint-HomeAssistant#send-gcode-from-lovelace ) then the Anycubic Kobra printer perfectly fulfills the command and does not reset the coordinates and does not signal with a sound signal, and if I use the MQTT Subscribe plugin and specify the code `{ "command": "G1 Y200 F3000" }`, then the printer loses coordinates, stops working and a lot it signals multiple times with an audible signal.  The problem comes out in your MQTT Subscribe plugin, which causes an error and the printer stops working at all, only a full reboot brings it to life

```
type: entity-button
name: Level Bed
tap_action:
  action: call-service
  service: mqtt.publish
  service_data:
    topic: octoPrint/hassControl/commands
    payload: "G29"
```

С плагином OctoPrint-HomeAssistant работает отлично
[image]

```
service: mqtt.publish
data:
  topic: octoPrint/hassControl/commands
  payload: "\"G1 Y200 F3000\""

```

jneilliii, 2022-12-24:
Enable debug logging for the plugin in OctoPrint's logging section and attempt the command then share your log. Need to check for errors.

jneilliii, 2022-12-24:
This worked for me...

[image]

[image]

[image]

jneilliii, 2022-12-24:
Here's a more advanced example, extracting all the individual bits.

[image]

[image]

[image]

DivanX10, 2022-12-24:
> Enable debug logging for the plugin in OctoPrint's logging section and attempt the command then share your log. Need to check for errors.

Thanks. Did I understand you correctly that you need to select octoprint.plugins.mqttsubscribe and error for logging?
[image]

jneilliii, 2022-12-25:
Yeah, but instead of error, change the drop down to debug. Either way, I think my examples above should work for your needs.

DivanX10, 2022-12-25:
Checked it out. All options have worked, except for one option, if I specify one axis instead of 3 axes. I understand that you need to specify a topic like MQTT. I was surprised at first that you specified the topic `octoPrint`, since I thought that MQTT Subscribe looks at MQTT regardless of the topic `octoPrint`, listens to it, and as it sees its topic, which we specify in the line, it reacts and sends a command to the printer.  And yet, I specified the topic `octoprint`, and the MQTT plugin specified the topic `octoPrint` and apparently, due to the fact that the register check is turned off, hence the heated platform options and nozzles worked on the topic `octoprint`. Therefore, according to your examples, it didn't work for me right away, until I paid attention to which topic you indicated, and as the topic `octoPrint` indicated, it worked.

This working option

**Topic:** `octoPrint/cmd/command`
**JSONPath Extract:** `($.x)|($.y)|($.z)|($.speed)`
**REST API:** `/api/printer/command`
**REST Parameters:** `{"command":"G1 X{0} Y{1} Z{2} F{3}"}`

Starting the service in Home Assistant
```
service: mqtt.publish
data:
  topic: octoPrint/cmd/command
  payload: '{"x":50,"y":50,"z":20,"speed":1000}'
``` 
-------
It does not work for one axis, probably it is not provided. How can this be implemented for one axis or only for two axes?

**Topic:** `octoPrint/cmd/command`
**JSONPath Extract:** `($.y)|($.speed)`
**REST API:** `/api/printer/command`
**REST Parameters:** `{"command":"G1 Y{1} F{3}"}`

Starting the service in Home Assistant
```
service: mqtt.publish
data:
  topic: octoprint/cmd/command
  payload: '{"y":50,"speed":2000}'
```
------

This option also worked

**Topic:** `octoPrint/cmd/command`
**JSONPath Extract:** `$.gcode`
**REST API:** `/api/printer/command`
**REST Parameters:** `{"command":"{0}"}`

Starting the service in Home Assistant
service: mqtt.publish
data:
  topic: octoPrint/cmd/command
  payload: '{"gcode":"G1 Y50 F3000"}'

-----

These are working examples
[image]
[image]
[image]
[image]
[image]

DivanX10, 2022-12-25:
> Yeah, but instead of error, change the drop down to debug. Either way, I think my examples above should work for your needs.

I have something not writing to the log, although I chose the octoprint.plugins.mqttsubscribe debug mode. How many errors I have surfaced, but there are no entries in the logs
[image]

DivanX10, 2022-12-25:
Thank you very much, I'm closing the topic

closed 2022-12-25 by DivanX10

jneilliii, 2022-12-25:
the plugin will write it's info into octoprint.log. I've never thought of a way to do optional parameters being sent. not sure if there is a way to do that in the current implementation. would have to experiment a bit. if I convert the plugin to using python f string conversion I might be able to do some more complicated replacements.

jneilliii, 2022-12-25:
> since I thought that MQTT Subscribe looks at MQTT regardless of the topic `octoPrint`

This is true. It doesn't matter what the topic is. For example in this screenshot, both of these topics do the same thing.

[image]